robertzaage/GroBro #1: Noah 2000 integration

joni347, 2025-03-29:
Would be cool if you also integrated the Noah 2000!

joni347, 2025-03-29:
Excuse me, but how can I send you an email?

joni347, 2025-03-29:
I really don't know how to contact you by email as I can't find your email address anywhere. If you are really interested in me connecting my Noah to your mqtt server, you can contact me at joni777 at web.de.

robertzaage, 2025-03-30:
> I really don't know how to contact you by email as I can't find your email address anywhere. If you are really interested in me connecting my Noah to your mqtt server, you can contact me at joni777 at web.de.

If you clone a repo the commit author mail is visible in git log. I will setup another MQTT instance and send you an email later.

joni347, 2025-03-30:
I'm pretty new to github, thanks for the tip!

robertzaage, 2025-03-30:
> I'm pretty new to github, thanks for the tip!

I sent you a message with instructions. 😊