robertzaage/GroBro #3: Data collection only?

wastlbava, 2025-03-30:
Hi,
I came across your interesting project while searching for inverters and power storage units with local APIs.

Is this integration only intended to retrieve data for display in Home Assistant or can it also be used to change values on the inverter? (For example, changing the output power of the inverter)

Unfortunately, I'm not that familiar with this topic.

robertzaage, 2025-03-30:
Currently it's read-only. But in the future controlling the device could be a viable option.

wastlbava, 2025-03-31:
Thank you very much for your feedback and your work on the integration.
It would be great if it were possible to control it in the future.

Does this mean that you currently set the values via the ShinePhone app via Bluetooth if necessary?

robertzaage, 2025-03-31:
> Thank you very much for your feedback and your work on the integration. It would be great if it were possible to control it in the future.
> 
> Does this mean that you currently set the values via the ShinePhone app via Bluetooth if necessary?

Yes.

closed 2025-03-31 by wastlbava

tobuh, 2025-04-21:
@robertzaage Have you ever succesfully written to mqtt topics? (e.g. for setting output power of NOAH). Could you elaborate a bit about writing to mqtt? I tried to find some information about the data used in the mqtt topics. (probably modbus encapsulated in mqtt, right?) But I could not find any resource. Happy about any guidance on this ... I am willing to help. 

My setup: NOAH with NEO800 and Shelly 3EM. Ideally, I could set up the "zero feed-in" (Nulleinspeisung) without Growatt cloud ...