robertzaage/GroBro #4: Zertifikate für lokalen Server

vandecook, 2025-04-06:
Hallo,
können die Zertifikate einer eigenen Domain auch verwendet werden, wenn ein rein lokaler Server verwendet wird, der nur über seine IP verwendet wird? 

Bei mir kann sich der Noah2000 verbinden, jedoch sind die Werte die ich ins HA bekomme nicht korrekt. ZB Pv Energy total [image]

Stimmen die links in der Beschreibung noch? Ich konnte die Zertifikate nicht darüber runterladen.

vandecook, 2025-04-06:
Aktuell habe ich eine Domain die über dyno auf auf meinen Reverse Proxy läuft. Darin habe ich für die Domain.tld ein Zertifikat erstellt und dieses für die lokale Instanz verwendet.

Die Verbindung hat zumindest funktioniert. 

Bei mir läuft ein MikroTik Router. Da müsste ich mir anschauen wie das umschreiben funktioniert.

Dotmaster5, 2025-04-13:
Hallöchen,
ich versuche es auch Lokal nur über IP hinzubekommen um die Werte dann im Homeassistant verarbeiten zu können, ich habe hier im Mehrfamilienhaus mehrere Neo 800 für die Balkonanlagen. 

Leider scheitere ich auch komplett an der Let’s Encrypt Zertifikat Geschichte. Ich habe bis jetzt noch nie etwas damit gemacht und bin kompletter Anfänger, werde aber leider auch aus der Anleitung nicht schlau und verzweifele daran. Falls von relevanz, ich benutze Opensense und einen MikroTik Switch. 

Könntet ihr mir vielleicht erklähren, wie ich das mit dem Let’s Encrypt Zertifikat, local zum laufen bekomme oder besser gesagt, wie ich mir das passende Zertifikat erstelle? Außerdem funktioniert der erste Link aus der Anleitung  ( wget https://letsencrypt.org/certs/lets-encrypt-e6.pem ) leider auch nicht mehr.

Gibt es nicht die Möglichkeit, das ganze auf Homeassistant als Addon auszuführen und es erstellt sich dann alle Zertifikate  und was sonnst noch so benötigt wird von selbst? 

Dies ist übrigens mein erster Post auf github und ich bitte um Entschuldigung, wenn ich vielleicht etwas falsch gemacht habe oder meine Frage etwas dumm klingt, aber ich versuche schon seit über einer Woche ohne Erfolg, die Sache mit Let’s Encrypt zum laufen zu bringen.

Beste Grüße und schon einmal 1000 x Danke :-)

ncwoehler, 2025-04-14:
Ich habe es lokal mit Mikrotik so hinbekommen:

1. Zertifikat manuell erstellen:
   ```
   sudo certbot certonly --manual --preferred-challenges dns -d <your-domain>
   ``` 

2. Root Zertifikat herausfinden:
   ```
   sudo awk 'BEGIN{n=0} /BEGIN CERT/{n++} {print > ("cert" n ".pem")}' /etc/letsencrypt/live/<your-domain>/fullchain.pem
   for i in cert*.pem; do   echo ">>> $i";   openssl x509 -in "$i" -noout -subject -issuer; done
   ```
   (isrgrootx1 bei mir)

3. Volles Zertifikat erstellen:

   ```
   curl -o root.pem https://letsencrypt.org/certs/isrgrootx1.pem
   sudo cat /etc/letsencrypt/live/<your-domain>/fullchain.pem root.pem > chain-full.pem
   ```

4. Mosquitto mit TLS starten, wie [hier](https://github.com/robertzaage/GroBro/blob/main/CONFIGURATION.md#2-mosquitto-tls-configuration) angegeben
 
5. Mikrotik DNS Eintrage für \<your-domain> auf \<mosquitto-tls-ip> erstellen:
  
   ```
   /ip dns set allow-remote-requests=yes
   /ip dns static add name="<your-domain>" address=<mosquitto-tls-ip> ttl=1d
   ```

6. Growatt Neo 800M-X auf \<your-domain> konfigurieren

Einziges Problem: Das erstellte Zertifikat muss anscheinend alle 3 Monate erneuert werden. Dazu muss ich mir noch eine Automatisierung einfallen lassen.

vandecook, 2025-04-14:
Danke. Das werde ich ausprobieren.
Aktuell habe ich einfach die Zertifikate von meiner Reverse Proxy Instanz genommen und hinkopiert. 

[untested] Mit dem DNS eintrag vermute ich das Problem, dass ich die ganze Domain umleite. Wäre besser eine Subdomain zu nehmen. oder? 
Aktuell habe ich einfach die IP vom MQTT server mit TLS angegeben. Das läuft bis jetzt hervorragend.

An das 3 monatliche Update habe ich nicht gedacht. Das könnte ich mit einem Cron Job machen, der mir die neuen Zertifiakte erstellt, kopiert und den MQTT Server neu startet.

ncwoehler, 2025-04-14:
Bei mir funktioniert die ganze Domain umzuleiten ohne Probleme. Semantisch wäre es vielleicht besser eine Subdomain zu nehmen.

Das Updater der Zertifikate braucht einen manuellen Eingriff, um den TXT DNS Record zu setzen. Daher könnte ein Cron Job schwierig werden. Aber vielleicht gibt es auch Anbieter von Domains, die eine API zum setzen von DNS Records anbieten. Dann kann man das ganze automatisieren. Darüber mache ich mir in 3 Monaten Gedanken, wenn der POC lang genug stabil lief 😉

vandecook, 2025-04-14:
Ich verwende DYNU. Dieser hat eine API zum setzen der DNS Records. Auf dem Router läuft ein Script, dass mit den DNS Record updatet.
Ich warte auch mal die Monate ab, so wie ich es jetzt verwendet und wenn es mit der IP und dem Zertifikat nicht mehr funktioniert stelle ich auf deine Lösung mit dem DNS Eintrag um.

closed 2025-04-25 by robertzaage