robertzaage/GroBro #5: Parall

vandecook, 2025-04-06:


robertzaage, 2025-04-06:
Dazu muss eine Art transparenter MQTT Proxy gebaut werden. Ich hatte noch keine Zeit, mir genauer anzuschauen wie man das am besten umsetzt.

Das Feature ist allerdings geplant, weil Growatt über einen Rückkanal bisher unbekannte Befehle an die Geräte schicken kann.

justinh998, 2025-04-06:
Dass Problem daran könnte aber natürlich sein, dass Growatt eventuell auch einen Befehl senden könte, der den MQTT Server im Gerät wieder zurückändert (wir wissen ja auch nicht wie Growatt intern mit dem Thema einer lokalen Steuerung umgeht ( Ist dass gewünscht? Entwicklen sie vieleicht sogar selbst gerade eine lokale Schnittstelle?)
Die Kommunikation des Noahs ist bisher sehr unerforscht.
Dass hier ist (zumindestens laut meiner Recherche) bisher dass einzige Projekt in dem irgendwas dazu dokumentiert ist.

robertzaage, 2025-04-06:
Ich persönlich habe nur einen NEO-Inverter und habe diesen derzeit komplett vom Internet getrennt. Die Integration in Home Assistant klappt wesentlich besser als durch die Growatt-Cloud. Bisher wurde nichts ohne mein Wissen zurückgesetzt oder verändert. Wenn der MQTT Proxy also auf Wunsch nur die Publish Richtung zulässt kann Growatt über den MQTT Kanal nichts anstellen.

vandecook, 2025-04-06:
Ich hatte mal mit einem Growatt Menschen Kontakt und auch die Idee einer lokalen API parallel zu deren API beschrieben. Egal ob Rest oder MQTT. Laut ihm gibt es intern keine Pläne. Er meinte aber, er gibt die Idee "weiter". 

Ich denke, wenn man am Noah was ändern will, dann wird das über MQTT retour gemacht.

justinh998, 2025-04-06:
Ich hab jetzt einfach mal ausprobiert, eine MQTT-Bridge mit try_private=false einzurichten. Das heißt, dem Growatt-Server wird nicht mitgeteilt, dass die Verbindung über einen gebridgten Broker läuft – mein Broker verhält sich also wie ein ganz normaler Client.

Zum Testen hab ich erstmal bridge_insecure=false gesetzt, weil ich nicht weiß, welches CA Growatt verwendet. Die Verbindung kommt tatsächlich zustande, wird aber nach etwa einer Sekunde vom Growatt-Server wieder gekappt.

robertzaage, 2025-04-06:
> Ich hab jetzt einfach mal ausprobiert, eine MQTT-Bridge(die alles weiterleitet , da auf diesen MQTT Broker sowieso nur Nachrichten vom Noah ankommen) mit try_private=false einzurichten. Das heißt, dem Growatt-Server wird nicht mitgeteilt, dass die Verbindung über einen gebridgten Broker läuft – mein Broker verhält sich also wie ein ganz normaler Client.
> 
> Zum Testen hab ich erstmal bridge_insecure=false gesetzt, weil ich nicht weiß, welches CA Growatt verwendet. Die Verbindung kommt tatsächlich zustande, wird aber nach etwa einer Sekunde vom Growatt-Server wieder gekappt.

Womit genau hast du das probiert? Wird die Client ID (Seriennummer) des NOAH weitergeleitet? Wird auch eine eventuell notwendige Authentifizierung durchgereicht?

Growatt hat ein DigiCert Zertifikat:
```
~ $ openssl s_client -showcerts -connect mqtt.growatt.com:7006
Connecting to 47.254.130.145
CONNECTED(00000003)
depth=2 C=US, O=DigiCert Inc, OU=www.digicert.com, CN=DigiCert Global Root G2
verify return:1
depth=1 C=US, O=DigiCert Inc, OU=www.digicert.com, CN=Encryption Everywhere DV TLS CA - G2
verify return:1
depth=0 CN=*.growatt.com
verify return:1
---
Certificate chain
 0 s:CN=*.growatt.com
   i:C=US, O=DigiCert Inc, OU=www.digicert.com, CN=Encryption Everywhere DV TLS CA - G2
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Feb 20 00:00:00 2025 GMT; NotAfter: Mar  1 23:59:59 2026 GMT
-----BEGIN CERTIFICATE-----
MIIF/jCCBOagAwIBAgIQCQlXy0umJkYxo27Q/dusDjANBgkqhkiG9w0BAQsFADBu
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMS0wKwYDVQQDEyRFbmNyeXB0aW9uIEV2ZXJ5d2hlcmUg
RFYgVExTIENBIC0gRzIwHhcNMjUwMjIwMDAwMDAwWhcNMjYwMzAxMjM1OTU5WjAY
MRYwFAYDVQQDDA0qLmdyb3dhdHQuY29tMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8A
MIIBCgKCAQEAqB6CZrkYrVHcXJZyGfIup8HM64NFuJeBsr0jAr9f9LDXYZjtu+2n
6ENWwfVaVRlHhSuLC65tDj9UnxnvClDqgu/JoK9Lj7TV5KGv+mYRsjiKQ7ykm0m/
f2e01gSTHxtbLPFAmNaR1AeVmbOp7ZrKeF9ulYu6RROFh58APmB8hyMkg0ja6MVq
cipadcIrgSCXDXIsoBZ8lF7yMNzmz0K6ICxGOvhubpZNv38IwaZhZXQ8ZRbWhRPM
NMyPSionE+K8W4N2PZJl+cdF27hBxBVv2wMSzw0aF2ys4ARX+m88+LVhFnaDhi0T
vdYGggBh0Rz9Ec498RfjgwdZ225S8fMC6QIDAQABo4IC7DCCAugwHwYDVR0jBBgw
FoAUeN+RkF/u3qz2xXXr1UxVU+8kSrYwHQYDVR0OBBYEFIR0KZ8ymFqiGBZpJzVh
FO761uzQMCUGA1UdEQQeMByCDSouZ3Jvd2F0dC5jb22CC2dyb3dhdHQuY29tMD4G
A1UdIAQ3MDUwMwYGZ4EMAQIBMCkwJwYIKwYBBQUHAgEWG2h0dHA6Ly93d3cuZGln
aWNlcnQuY29tL0NQUzAOBgNVHQ8BAf8EBAMCBaAwHQYDVR0lBBYwFAYIKwYBBQUH
AwEGCCsGAQUFBwMCMIGABggrBgEFBQcBAQR0MHIwJAYIKwYBBQUHMAGGGGh0dHA6
Ly9vY3NwLmRpZ2ljZXJ0LmNvbTBKBggrBgEFBQcwAoY+aHR0cDovL2NhY2VydHMu
ZGlnaWNlcnQuY29tL0VuY3J5cHRpb25FdmVyeXdoZXJlRFZUTFNDQS1HMi5jcnQw
DAYDVR0TAQH/BAIwADCCAX0GCisGAQQB1nkCBAIEggFtBIIBaQFnAHUADleUvPOu
qT4zGyyZB7P3kN+bwj1xMiXdIaklrGHFTiEAAAGVIQTj9QAABAMARjBEAiBB7TGC
cSqpyqh0wnXW/vjZZkAsCKb31Z4WIRU43/GQQQIgXLmE8byOhzv8Pt8x9/nZ+7Ep
dDxfzVqnogMyV3ikPVoAdQBkEcRspBLsp4kcogIuALyrTygH1B41J6vq/tUDyX3N
8AAAAZUhBOOxAAAEAwBGMEQCIFSwPr4C2FKR9CjHjj9NS7ELNlcoz+oDSL+7wGSd
FX3nAiBLxfyDr4Muhyk9AXq4D+KtR4RNhv6ZG9F1vPv5dnHfEwB3AEmcm2neHXzs
/DbezYdkprhbrwqHgBnRVVL76esp3fjDAAABlSEE48IAAAQDAEgwRgIhAJ9CQiCN
srVnUA37IgrgrMn1zpjaE6/q02s1B3G87GuWAiEAlKrmfhpeD5qXpMVE8K/s3phX
+NP9chiwT9SU6PzA7HgwDQYJKoZIhvcNAQELBQADggEBANxmXPJNP4nooDpVwOSR
GmSDVr5cEIqEMFjIGJBMbg1WCcYuIVZ/4cv33ElFI9VKUh4zgquIW7hdrRI1FFVg
+kzEJooAfJx5d1/IeSKAS5oKe9LyKbAOVs0qFN+G00gZmfDRWnBgR6QUjVV6HuAy
7XRunL9lbl3J4wtQGc2f31HGgPKGS6gPH/M8O+UOLM6xB7o0vf9f7+4kRp443ofs
oMTVwiKoHtFOgDGbhZ1aR8pV0V7ipV1bmjw73Fi8ArpEBybxKmIH6LTyyNx6FGBZ
Kq813OYw3Rh+Rrb/2HvrSuWdDJhwqlV929m5Cvn841x0HH9qnfuyHfPVgD0Be8Nm
Uks=
-----END CERTIFICATE-----
 1 s:C=US, O=DigiCert Inc, OU=www.digicert.com, CN=Encryption Everywhere DV TLS CA - G2
   i:C=US, O=DigiCert Inc, OU=www.digicert.com, CN=DigiCert Global Root G2
   a:PKEY: rsaEncryption, 2048 (bit); sigalg: RSA-SHA256
   v:NotBefore: Nov 27 12:46:40 2017 GMT; NotAfter: Nov 27 12:46:40 2027 GMT
-----BEGIN CERTIFICATE-----
MIIEqjCCA5KgAwIBAgIQDeD/te5iy2EQn2CMnO1e0zANBgkqhkiG9w0BAQsFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBH
MjAeFw0xNzExMjcxMjQ2NDBaFw0yNzExMjcxMjQ2NDBaMG4xCzAJBgNVBAYTAlVT
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j
b20xLTArBgNVBAMTJEVuY3J5cHRpb24gRXZlcnl3aGVyZSBEViBUTFMgQ0EgLSBH
MjCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAO8Uf46i/nr7pkgTDqnE
[... 71 lines cut ...]
    Cipher    : ECDHE-RSA-AES128-GCM-SHA256
    Session-ID: 1A503F13B90D56AC0489C0574F1CCE831F2DAA406DE9A06E2CB5560D7E427A0C
    Session-ID-ctx:
    Master-Key: 7BD6CAA24BCD1844AF9D3FBCEBD01B873F171F6A39EB9C0AFD351F50CF52DB18C4D5E2796598157769D291D6D57A5338
    PSK identity: None
    PSK identity hint: None
    TLS session ticket lifetime hint: 300 (seconds)
    TLS session ticket:
    0000 - 66 5d b1 3c 44 a2 e7 cf-74 9f 3a 4c 39 b1 f6 8f   f].<D...t.:L9...
    0010 - 45 17 ef c8 a2 10 23 de-b1 c2 d9 27 4f b3 55 f3   E.....#....'O.U.
    0020 - 37 19 66 6f 17 9c 91 d0-42 4f 90 04 16 0a 1b 3e   7.fo....BO.....>
    0030 - 11 b4 a4 a3 91 60 8f b6-62 59 55 4b 3f bb 56 c2   .....`..bYUK?.V.
    0040 - b9 1b 1c 12 b3 a9 ba 7a-aa d8 db f9 a4 80 7b 0c   .......z......{.
    0050 - 90 49 36 4d 6a e5 2d 69-a0 d7 2e b8 df 6c 8d e1   .I6Mj.-i.....l..
    0060 - 03 d0 ff 6a 4c 79 ba 65-17 c3 a4 18 00 26 07 cd   ...jLy.e.....&..
    0070 - 85 7f 00 01 12 4c 91 86-26 31 07 5a 33 40 8c b7   .....L..&1.Z3@..
    0080 - 8f db 1e 72 33 8a c2 3b-e4 05 74 ad 81 91 cd ed   ...r3..;..t.....
    0090 - c7 48 84 c6 7f ed be d9-2d 36 89 f0 11 1e 28 12   .H......-6....(.
    00a0 - dd b5 f7 0d 83 63 7d 00-59 44 59 36 9e 9d d4 cf   .....c}.YDY6....

    Start Time: 1743964190
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: yes
```

justinh998, 2025-04-07:
Ich habe das Ganze in meinem MQTT-Broker eingerichtet, der die Nachrichten vom Akku empfängt.

Ich habe den Code  von GroBro jetzt aber sowieso [angepasst](https://github.com/robertzaage/GroBro/commit/9ed869440c9c96ad7b56ef5740a81e6e1516d697), um das direkt dort umzusetzen ,aus meiner Sicht die einfachere Lösung. Das ganze hat sogar funktioniert. Zumindest für ca. 10 Minuten, danach war der Akku wieder offline. Die Befehle wurden aber weiterhin weitergeleitet.

Ich schätze mal, dass der Growatt-Server mehrfach versucht, Daten an Noah zu senden, und wenn das nicht klappt, wird der Akku als offline eingestuft.  
Theoretisch sollte das Empfangen und Weiterleiten  von Nachrichten mit der Modifikation in beide Richtungen funktionieren, aber ich hatte noch keine Zeit, das noch zu testen.

justinh998, 2025-04-07:
Kleine Ergänzung zu oben.
Der Growatt Server sendet auf dem Topic
s/(Srn des Gerätes) Daten. Der Noah hat in meinem Fall aber nur dass Topic
s/33/(Srn des Gerätes) abonniert.
Theoretisch könnte das Topic vor dem Weiterleiten umgeschrieben werden ( ich weiß aber nicht ob die 33 zufällig gewählt wird) aber eigentlich sollte die Kommunikation auch so funktionieren, dass ganze soll nur als Transparenter MQTT Broker dienen,der keine Daten verändert.

robertzaage, 2025-04-07:
> Kleine Ergänzung zu oben.
> Der Growatt Server sendet auf dem Topic
> s/(Srn des Gerätes) Daten. Der Noah hat in meinem Fall aber nur dass Topic
> s/33/(Srn des Gerätes) abonniert.
> Theoretisch könnte das [...]

Die 33 scheint bei allen NEO und NOAH Geräten   fest eingestellt zu sein. Kannst du die Growatt API Messages dumpen?

justinh998, 2025-04-07:
Okay, ich finde es ehrlich gesagt trotzdem etwas merkwürdig, weil es ohne einen transparenten Proxy auch ohne Topic-Manipulation funktioniert. Ich habe das jetzt statisch auf s/33/(Srn des Gerätes) geändert und jetzt bleibt der Akku auch online und ich kann ihn auch über die ShinePhone App / die Growatt API steuern.

Ich werde die heute Abend mal dumpen (dann kann ich auch gleich Steuerbefehle wie das Setzen der Standard-Systemausgangsleistung dumpen).

Ergänzung:
Ich weiß nicht, ob Growatt aktuell Probleme hat oder ob es an der Weiterleitung liegt, aber derzeit sind die Daten in der Growatt Cloud extrem verzögert. Ich frage die Growatt API alle 5 Sekunden ab. Gestern, als der Akku noch verbunden war, hat er alle Änderungen sehr schnell angezeigt. Wenn ich z. B. die System Output Power auf 800 W hochgestellt habe, hat er in der Regel schon nach 5 Sekunden veränderte Werte für die Akku-Lade-/Entlade-Power angezeigt. Jetzt dauert das fast 30 Sekunden, bis sich da was verändert, obwohl er laut WR schon längst die Leistung erhöht hat. Dass Erhöhen der Leistung an sich verzögert sich aber anscheinend nicht.
Die ShinePhone App funktioniert aktuell auch überhaupt nicht (sie findet nicht mal den Akku als Gerät).
Ich wüsste aber auch nicht, wie durch das Tool eine riesige Verzögerung entstehen soll.
Der Akku sendet außerdem auch nur alle 5 Sekunden an den Broker. Als er noch direkt verbunden war, hat er, zumindest laut Netzwerk-Mitschnitt, jede 1-2 Sekunden gesendet.Eventuell führt dass ja zu dieser riesen Verzögerung.
Hat noch jemand anderes mit einem Akku, der noch direkt mit dem Server verbunden ist, diese Probleme?

robertzaage, 2025-04-07:
> Okay, ich finde es ehrlich gesagt trotzdem etwas merkwürdig, weil es ohne einen transparenten Proxy auch ohne Topic-Manipulation funktioniert. Ich habe das jetzt statisch auf s/33/(Srn des Gerätes) geändert und jetzt bleibt der Akku auch online und ich kann ihn auch über die ShinePhone App / die Growatt API steuern.
> 
> Ich werde die heute Abend mal dumpen (dann kann ich auch gleich Steuerbefehle wie das Setzen der Standard-Systemausgangsleistung dumpen).
> 
> Ergänzung:
> Ich weiß nicht, ob Growatt aktuell Probleme hat oder ob es an der Weiterleitung liegt, aber derzeit sind die Daten in der Growatt Cloud extrem verzögert. Ich frage die Growatt API alle 5 Sekunden ab. Gestern, als der Akku noch verbunden war, hat er alle Änderungen sehr schnell angezeigt. Wenn ich z. B. die System Output Power auf 800 W hochgestellt habe, hat er in der Regel schon nach 5 Sekunden veränderte Werte für die Akku-Lade-/Entlade-Power angezeigt. Jetzt dauert das fast 30 Sekunden, bis sich da was verändert, obwohl er laut Wechselrichter schon längst die Leistung erhöht hat. Dass Erhöhen der Leistung an sich verzögert sich aber anscheinend nicht.
> Die ShinePhone App funktioniert aktuell auch überhaupt nicht (sie findet nicht mal den Akku als Gerät).
> Ich wüsste aber auch nicht, wie durch das Tool eine riesige Verzögerung entstehen soll.
> Der Akku sendet außerdem auch nur alle 5 Sekunden an den Broker. Als er noch direkt verbunden war, hat er, zumindest laut Netzwerk-Mitschnitt, jede 1-2 Sekunden gesendet.Eventuell führt dass ja zu dieser riesen Verzögerung.
> Hat noch jemand anderes mit einem Akku, der noch direkt mit dem Server verbunden ist, diese Probleme?

Erstell gerne einen PR mit deinen Änderungen. Die Steuerbefehle sind am spannendsten, da ich bisher keine analysiert habe. Magst du mir deinen Dump per Mail zukommen lassen (steht im git log des repo).

vandecook, 2025-04-07:
Ich habe einen Speicher und sollte heute zwei weitere bekommen. Ich überlege einen über GroBro zu betreiben. 
Ich würde einen Dump liefern. Sag mir einfach, wie ich dazu komme.

justinh998, 2025-04-08:
Kleines Update. Seit gestern Mittag läuft gar nichts mehr . Eigentlich habe ich dass System auf dem dass ganze läuft einfach nur neu gestartet aber seitdem ist der Akku im Growatt System dauerhaft offline.

Die Daten werden aber weiterhin an den Growatt MQTT Server durch geleitet und es wurden sogar seit gestern Mittag 3 Steuerungs Anfragen von der Cloud empfangen, die auch weitergeleitet wurde.

Ich werde heute Abend / morgen Abend mal versuchen den Akku mal neu einrichten. Ich habe die Hoffnung,dass beim Einrichten eventuell Informationen gesendet werden, die wichtig für die Growatt Cloud sind.

In diesen Zustand werde ich auf jeden Fall kein MR stellen , dass ganze läuft viel zu instabil.

joni347, 2025-04-08:
@justinh998 momentan hat growatt aber auch massiv probleme mit der cloud, nicht das es daran liegt.
https://akkudoktor.net/t/growatt-noah-2000-erfahrungen/16700/2207?page=39

joni347, 2025-04-08:
@robertzaage könnte es sein das die /33/ eventuel in hex und ASCII ist? Also 0x33 -> '3'. Das würde für die 03 Hold register aus dem Modbus protokoll sprechen.

justinh998, 2025-04-08:
> [@justinh998](https://github.com/justinh998) momentan hat growatt aber auch massiv probleme mit der cloud, nicht das es daran liegt. https://akkudoktor.net/t/growatt-noah-2000-erfahrungen/16700/2207?page=39

Dass könnte eventuell sogar sein. Die Grôßten Probleme gab es anscheinend vor 4-6 Tagen. Da hatte ich den Noah tatsächlich auch noch in der Growatt Cloud und er war dauerhaft online.

justinh998, 2025-04-09:
Erneutes Update. Heute habe ich es zu mindestens geschafft den Noah online zu bringen aber nicht lange. Er wurde für ca 1 Min als online angezeigt und  die Werte in der Growatt Cloud haben sich innerhalb dieser Minute auch nur einmal geupdatet.

justinh998, 2025-04-10:
Nach dem Neueinrichten des Noahs funktioniert es aktuell sehr gut, sowohl dass Steuern als auch dass reine Auslesen der Daten von der Cloud.🎉🎉. 
Ich bin mir zwar nicht sicher, ob das Ganze auch einen Neustart des Servers oder einen Disconnect zwischen dem Akku und meinem Server bzw. zwischen meinem Server und dem Growatt-Server übersteht, aber das teste ich später noch. Nach dem Neueinrichten des Noahs wird auch direkt an das Topic s/33/{GeräteSrn} gesendet, sodass ein manuelles Republishing nicht mehr notwendig ist.

joni347, 2025-04-10:
@justinh998 das hört sich richtig gut an! Musste meinen Noah die letzten 2 Wochen auch sehr oft neu einrichten. Die beste Abhilfe war bis jetzt das ich ihn aus dem AVM Mesh genommen und einen extra Accesspoint 1 Meter neben ihm eingerichtet habe. Jetzt wäre interessant auf welchen topics der growatt server sendet wenn man Einstellung in der App verändert. Das würde @robertzaage bestimmt viel helfen, er hat auch schon einige topics entschlüsselt.

justinh998, 2025-04-10:
Leider sendet der Growatt Server nach einem Neustart meines Systems wieder auf s/{SrndesAkkus} . Wir werden die Weiterleitung also leider doch brauchen.Zumindestens scheint dass ganze auch nach einer längeren Unterbrechnung normal weiterzulaufen.

joni347, 2025-04-10:
> [@joni347](https://github.com/joni347) Ja dass herauslösen aus dem mesh hat bei mir auch geholfen. Der Akku hat sich aber auch immer mit dem am weitesten entfernten Accespoint verbunden 😒. Ich habe Robert gerade eine E-Mail mit den Dumps geschickt. Ich würde sie hier auch veröffentlichen und eigentlich bin ich auch der meinung dass dort nichts wirklich persönliches drinnen steht aber Vorsicht ist besser als Nachsicht. Wer die Dumps haben will kann mir aber gerne eine PN schreiben oder sich selbst welche erzeugen ( ich werde gleich noch meinen jetzigen Code Stand in meinen Fork pushen).

Die SrNr deines Akkus steht auf jeden Fall in den Nachrichten nochmal drin, also lieber nur per PN ;)

robertzaage, 2025-04-12:
Wurde mit v1.4 implementiert.