robertzaage/GroBro #8: Werte von Noah (2 Stück als Turm) nicht korrekt

vandecook, 2025-04-12:
Verwendete Version
- 1.4

Setup 
- 2 Speicher NAOH2000 als Turm
- SN der Kopfspeichers als environment Variable angegeben

FW Noah 2000
- 10.09.05.4013

Daten werden an HomeAssistant übergeben.
Einige Werte in HomeAssistant sind zuordenbar, manche nicht. Manche würden sich erklären lassen.

Ergebnis und meine Interpretation in der Graphik
[image]

Rohdaten können bei Bedarf zur Verfügung gestellt werden.

robertzaage, 2025-04-12:
Vielen Dank für den Hinweis, das NOAH Mapping ist stellenweise noch WIP! Kannst du mir die Rohdaten per Mail schicken? (steht im git log des repos)

Besonders spannend an deinem Setup ist die Frage ob nur "der Kopf" Daten sendet oder beide unabhängig voneinander. Das wirkt sich sicherlich auch auf das Mapping aus.

vandecook, 2025-04-12:
Die Rohdaten bekomme ich vom MQTT server?  
/c/33/SN****/ 
Der String der da drinnen steht?

vandecook, 2025-04-12:
Leider bin ich etwas zu "dumm" um zu deiner Emailadresse zu kommen.

robertzaage, 2025-04-12:
> Die Rohdaten bekomme ich vom MQTT server? /c/33/SN****/ Der String der da drinnen steht?

Genau dahin wird gepublished. Meine mail ist robert [at] zaage.it

robertzaage, 2025-04-12:
Ich denke wir brauchen für sowas eine Dump Funktion in GroBro. Das wäre dann wesentlich einfacher für dich. 🙂

xdatravelbug, 2025-04-17:
Not sure if that helps, but I read all possible registers for my Noah2000 from the Growatt online account, trying to figure out if there is anything else interesting in the data:
[Noah2000_Register.xlsx](https://github.com/user-attachments/files/19793290/Noah2000_Register.xlsx)

robertzaage, 2025-04-19:
> Not sure if that helps, but I read all possible registers for my Noah2000 from the Growatt online account, trying to figure out if there is anything else interesting in the data: [Noah2000_Register.xlsx](https://github.com/user-attachments/files/19793290/Noah2000_Register.xlsx)

Thank you very much! 😊

vandecook, 2025-04-22:
> Not sure if that helps, but I read all possible registers for my Noah2000 from the Growatt online account, trying to figure out if there is anything else interesting in the data: [Noah2000_Register.xlsx](https://github.com/user-attachments/files/19793290/Noah2000_Register.xlsx)

Where to get the data for the registers from the Growatt account?

vandecook, 2025-04-22:
@robertzaage do you have a new version for testing? Or is there a possilibty to try to help to decode the data?

robertzaage, 2025-04-23:
> > [@robertzaage](https://github.com/robertzaage) do you have a new version for testing? Or is there a possilibty to try to help to decode the data?
> 
> Soon... Another helpful user managed to interprete some additional voltage registers. I will check this later on my historic dataset and update the mapping. For bleeding edge changes I created the `:dev` image tag last week.

@vandecook please try the new `dev` version

vandecook, 2025-04-23:
Thanks. I will do this.

vandecook, 2025-04-23:
Now working with :dev version.  A big step forward. Thanks a lot.

Some data are wrong. E.g. or the naming is incorrect

- Battery 2 SoC: value is different to value in app
- Energy output today ->Total Power Generation (of head storage)
- PV energy total -> should be Monthly Power Generation
- Energy output total -> is Energy output total from all connected batteries
- Cell 1 voltage: unclear
- Cell 2 voltage: unclear
- Cell 3 voltage: unclear
- Cell 4 voltage: unclear
- PV1 voltage: NOK
- PV2 voltage: NOK
- PV1 input current: must be checked during the day
- PV2 input current: must be checked during the day
- Temperature: NOK

robertzaage, 2025-04-24:
> Now working with :dev version. A big step forward. Thanks a lot.
>
> Some data are wrong. E.g. or the naming is incorrect
>
>     * Battery 2 SoC: value is different to value in app
>
>     * Energy output [...]

I pushed another update tonight. Please keep in mind to remove the device or entities from HA, as most of the sensor names changed after fiddling out what the values really meant to be.

whizzdix, 2025-04-24:
hello all,
the project is really great and you are making very good progress. thanks for that!

i noticed the following with the noah data, tested with two different noahs and the last :env version:

1. "Energy output today" corresponds to my solar production this year
2. "PV energy total" corresponds to my solar production in the current month (April)
3. "PV energy today" corresponds to today's solar production (correctly implemented)
4. "Energy output total" corresponds to the total solar production of this noah (correctly implemented, name does not match)
5. "Output power limit", implemented in the last :dev release, is the current output power for me ("output power" no longer changes)

maybe it will help you.

robertzaage, 2025-04-25:
@whizzdix This helps a lot! I just updated the dev version with your corrections.

vandecook, 2025-04-25:
Register No 12 seems to be the number of batteries
- System 1: 1 Battery => Register No 12 = 1
- System 2: 2 Batteries => Register No 12 = 2

Currently I don´t want to change my system, but for me this makes sense.

robertzaage, 2025-04-25:
@vandecook Alright! I just pushed a fix.

msmthng, 2025-04-25:
Which values are currently supported? For my automation i am onöy using soc Overall and soc for single Batterien and current pv Power.

Since i can not run grobro in parallel with the cloud i am depending at least on Overall soc and current Power since this is used for automation to Controll the Inverters.

whizzdix, 2025-04-25:
@robertzaage: with pleasure :) 

Now it fits with one exception: "Energy output total" ("register_no": 76) is actually solar production this year ("PV energy year"). my last message was perhaps misleading.

By the way: If i take the mqtt messages with the topic s/33/<noah_sn> and publish them again with this topic, i can make the change again at a later time. since i can't decode the modbus messages yet, an easy way to make changes without the cloud :)

vandecook, 2025-04-25:
You can use it in parallel. I currently use it parallel to wait for a new Firmware to solve my MPPT issue on both systems

robertzaage, 2025-04-25:
> [@robertzaage](https://github.com/robertzaage): with pleasure :)
>
> Now it fits with one exception: "Energy output total" ("register_no": 76) is actually solar production this year ("PV energy year"). [...]

Fixed now. 😊 

I'm working on a CLI tool to send configuration messages and register settings. But I need to do more research on the underlying data format before I can publish it here.

Fun fact: Growatt sends the current date and time every few hours in a special message to keep the device clock in sync.

robertzaage, 2025-04-25:
> Which values are currently supported? For my automation i am onöy using soc Overall and soc for single Batterien and current pv Power.
> 
> Since i can not run grobro in parallel with the cloud i am depending at least on Overall soc and current Power since this is used for automation to Controll the Inverters.
> 
> Soon this is available i World move my 3 Batteries Tower to grobro and would be able to support debbuging.

I think GroBro should support all metrics youre looking for and the mentioned "forwarding mode" is working flawlessly too. 🙂

### Battery Metrics
| Variable Name      | Description             | Unit | Notes / Options            |
|:-------------------|:-------------------------|:----:|:---------------------------|
| bat_sysstate        | Battery System State     |      | Idle, Charging, Discharging |
| tot_bat_soc_pct     | Total Battery SoC        | %    |                           |
| vbat1               | Battery 1 Voltage        | V    |                           |
| vbat2               | Battery 2 Voltage        | V    |                           |
| bat_1_soc_pct       | Battery 1 SoC            | %    |                           |
| discharge_limit     | Discharge Limit          | %    |                           |
| charge_limit        | Charge Limit             | %    |                           |
| bat_cnt             | Battery Count            | V    |                           |
| bat_cyclecnt        | Battery Cycle Count      |      |                           |

### PV (Solar) Metrics
| Variable Name      | Description             | Unit | Notes / Options            |
|:-------------------|:-------------------------|:----:|:---------------------------|
| Vpv1                | PV1 Voltage              | V    |                           |
| Vpv2                | PV2 Voltage              | V    |                           |
| Ipv1                | PV1 Input Current        | A    |                           |
| Ipv2                | PV2 Input Current        | A    |                           |
| pv_tot_power        | PV Total Power           | W    |                           |
| pv_eng_today        | PV Energy Today          | kWh  |                           |
| pv_eng_month        | PV Energy Month          | kWh  |                           |
| pv_eng_year         | PV Energy Year           | kWh  |                           |

### Output Metrics
| Variable Name      | Description             | Unit | Notes / Options            |
|:-------------------|:-------------------------|:----:|:---------------------------|
| out_power           | Output Power             | W    |                           |
| out_power_limit     | Output Power Limit       | W    |                           |
| out_voltage         | Output Voltage           | V    |                           |
| eng_out_device      | Energy Output Device     | kWh  |                           |

### Temperature Metrics
| Variable Name      | Description             | Unit | Notes / Options            |
|:-------------------|:-------------------------|:----:|:---------------------------|
| temp2               | Temperature 2            | °C   |                           |
| temp1               | Temperature 1            | °C   |                           |

### Device Info
| Variable Name      | Description             | Unit | Notes / Options            |
|:-------------------|:-------------------------|:----:|:---------------------------|
| priority_mode       | Priority Mode            |      | Load First, Battery First, Grid First |
| ser_part_1          | Serial Part 1            | s    |                           |
| ser_part_2          | Serial Part 2            | s    |                           |
| ser_part_3          | Serial Part 3            | s    |                           |
| ser_part_4          | Serial Part 4            | s    |

whizzdix, 2025-04-25:
#

> Fixed now. 😊
>
> I'm working on a CLI tool to send configuration messages and register settings. But I need to do more research on the underlying data format before I can publish it here.
>
> Fun fact: [...]

Thank you. It works well :)

I'm looking forward to the feature of being able to make configuration messages and register settings without the cloud. That would be really great.

> "working flawlessly too" 
thats right :) The bridge has been working for me since the initial installation without any problems, not even with the bidirectional passthrough to the Growatt servers.

Have you been able to identify the "charge power" or "discharge power" in the registers?

Thanks for the great work!

robertzaage, 2025-04-26:
@whizzdix I just pushed a tool to read most of these config messages published on /s/33/<device_id> and the replies on register reads on /c/33/<device_id>. We're very close now sending our first self crafted messages 😄 

Here are some examples from a NOAH.
```json
{
  "msg_len": 46,
  "msg_type": 272,
  "device_id": "0PVPXXXXXXXX",
  "payload": {
    "action": "slot_create",
    "slot": 13,
    "start": "18:10",
    "end": "23:12",
    "power": 200
  }
}
{
  "msg_len": 40,
  "msg_type": 272,
  "device_id": "0PVPXXXXXXXX",
  "payload": {
    "action": "charge_limit",
    "upper": 90,
    "lower": 20
  }
}
{
  "msg_len": 36,
  "msg_type": 262,
  "device_id": "0PVPXXXXXXXX",
  "payload": {
    "subtype": "output_limit",
    "power": 600
  }
}

```

> Have you been able to identify the "charge power" or "discharge power" in the registers?
Not yet.

vandecook, 2025-04-26:
I read out futher adresses .. up to adress 400 and got some values. 
[Noah2000_Register_400.xlsx](https://github.com/user-attachments/files/19925205/Noah2000_Register_400.xlsx)

xdatravelbug, 2025-04-28:
Pls excuse the delayed response as I was on vacation :-)

@vandecook Yes, it was a manual process for me - register after register.... Thanks a lot for the additional reads! I stopped previously because I did not see any further information from higher registers.

Depending on how Growatt implemented it I expected a maximum of 256 registers, which equals 0xFF in hex.

vandecook, 2025-04-28:
> Pls excuse the delayed response as I was on vacation :-)
>
> @vandecook Yes, it was a manual process for me - register after register.... Thanks a lot for the additional reads! I stopped previously [...]

I guess there are still more Adresses. I will try further readouts after my garden is completed.

whizzdix, 2025-04-28:
i have adapted the grocli.py implementation and created a new version with the name gromqtt.py, so that the actions can be triggered as json via a target mqtt host (topic: grobro/cmd):
{
 "action": "charge_limit",
 "device_id": "xxxxxxxxxxxxx",
 "upper": 100,
 "lower": 10
}
if you are interested i can upload it. i just don't know my way around github yet...

robertzaage, 2025-04-28:
> i have adapted the grocli.py implementation and created a new version with the name gromqtt.py, so that the actions can be triggered as json via a target mqtt host (topic: grobro/cmd): { "action": [...]

Yes, please 😊

msmthng, 2025-04-29:
i have changed the mqtt server to gromqtt.mydomain, i have alse created a letsencrypt certifiacte for this host including the root. in the mosquitto tls log i see only the bridge is connected. how long it usually takes from changing the server host until the noah is connecting to that mqtt broker?

whizzdix, 2025-04-29:
> Yes, please 😊

Done 😊 Pull Request initiated

msmthng, 2025-04-29:
> > i have changed the mqtt server to gromqtt.mydomain, i have alse created a letsencrypt certifiacte for this host including the root. in the mosquitto tls log i see only the bridge is connected. how long it usually takes from changing the server host until the noah is connecting to that mqtt broker?
> 
> Only a few seconds after changing the mqtt host.
> 
> Please check whether the shineapp shows a current status. If so, then the change to the mqtt host did not happen.
> 
> How did you change it? It is also possible to change the register 17 directly on server.growatt.com (datalogger setting) and save the mqtt host there.

i said add new device on my shine app, selected the connected noah changed to custom config and entered the mqtt host with saving the data. if i check shine app and settings for my noah it has the right mqtt server.

how to change the datalogger setting in the webiste?

msmthng, 2025-04-29:
[image]
ist it domain and port?

msmthng, 2025-04-29:
[image]
state os always an empty json

msmthng, 2025-04-29:
[1745958679368.zip](https://github.com/user-attachments/files/19967090/1745958679368.zip)
bin message comming in from noah

robertzaage, 2025-04-29:
> <img alt="Image" width="478" src="https://private-user-images.githubusercontent.com/40269445/438943532-a7f35345-6b9f-4699-8638-6374dac01f25.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU5NjExMzAsIm5iZiI6MTc0NTk2MDgzMCwicGF0aCI6Ii80MDI2OTQ0NS80Mzg5NDM1MzItYTdmMzUzNDUtNmI5Zi00Njk5LTg2MzgtNjM3NGRhYzAxZjI1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDI5VDIxMDcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE2YmJhMWZlZjhmMTA2YzBiNzhkN2JmMjAzOTA1YzQzNjdhZDI4ODJiMjM5ZTM0N2FiZDUyZTdmYWQ3YTZiZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zZGBzl_m0K70vATzd2VDI5cpjWvOcwV6t4iQiGkVh9Q">
> 
> 2025-04-29 19:40:47,710 [INFO] Published state for 0xxxxxxxxxxxxxxxxxxxJT with 0 registers 2025-04-29 19:40:52,725 [INFO] Device 0xxxxxxxxxxxxxxxxxxxJT matched 0 registers after filtering. 2025-04-29 19:40:52,725 [INFO] Published state for 0xxxxxxxxxxxxxxxxxxxJT with 0 registers 2025-04-29 19:40:57,739 [INFO] Device 0xxxxxxxxxxxxxxxxxxxJT matched 0 registers after filtering. 2025-04-29 19:40:57,740 [INFO] Published state for 0xxxxxxxxxxxxxxxxxxxJT with 0 registers
> 
> did i miss something in the configuration?
> <img alt="Image" width="611" src="https://private-user-images.githubusercontent.com/40269445/438950696-e3913ac8-a1d4-459a-8b79-27e5c17563e1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU5NjExMzAsIm5iZiI6MTc0NTk2MDgzMCwicGF0aCI6Ii80MDI2OTQ0NS80Mzg5NTA2OTYtZTM5MTNhYzgtYTFkNC00NTlhLThiNzktMjdlNWMxNzU2M2UxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDI5VDIxMDcxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU4NzFkODBiM2M1ZjcyNTk4NGI0OTc4MWNmYWE1MmQ0ZmU4Y2JlYzAzZWY5OTZkNzNiOWJhNzJlNTNiM2RlOWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7smnbSf6ps4ttEiR2JNop7ecCxNKAciz_u_nuKkfOeA"> is the message commnig to the tls broker fine?

Yes, I think you didnt set the `REGISTER_FILTER` option. Auto-detection needs a fix.

msmthng, 2025-04-30:
i can not find the section where register 21 23 25 27 got converted to strings / ascii

msmthng, 2025-04-30:
Register for Serialnumbers:

BAT1 21 23 25 27
BAT2 33 35 37 39
BAT3 45 47 49 51
BAT4 57 59 61 63

again distance is 12 as for soc (maybe this helpts, having a value for batn the battery n + 1 might have the register no + 12

robertzaage, 2025-05-01:
> Register for Serialnumbers:
> 
> BAT1 21 23 25 27 BAT2 33 35 37 39 BAT3 45 47 49 51 BAT4 57 59 61 63
> 
> again distance is 12 as for soc (maybe this helpts, having a value for batn the battery n + 1 might have the register no + 12
> 
> Register 20 22 24 26 28 are also containing the serial for BAT 1 in format CC CCCC CCCC CCCC CC and BAT 2 to 4 starting with register no + 12

Are we talking about the 0PVPXXXXX serial numbers here?

robertzaage, 2025-05-02:
> Seems Register 99 & 100 contain the min and max Cell Voltage of Bat1 in mV. They are matching the Values which are received via the Growatt IOBroker Adapter.
> 
> Only issue, seems only 2 Digits after the , are send via Mqtt. Anything else to be changed to send 3 digits?
> 
> MQTT contains:
> 
> `maxcvbat1: 3.29 mincvbat1: 3.29`
> 
> { "register_no": 99, "variable_name": "maxcvbat1", "description": "Maximum Cell Voltage", "unit": "V", "size": 2, "multiplier": 0.001, "ha": { "name": "Battery 1 maxCellVoltage", "unit_of_measurement": "V", "icon": "mdi:flash", "state_class": "measurement", "device_class": "voltage" } }, { "register_no": 100, "variable_name": "mincvbat1", "description": "Minimum Cell Voltage", "unit": "V", "size": 2, "multiplier": 0.001, "ha": { "name": "Battery 1 minCellvoltage", "unit_of_measurement": "V", "icon": "mdi:flash", "state_class": "measurement", "device_class": "voltage" }
> 
> Edit: Changed line 189 now it is fine. ` reg["name"]: round(reg["value"], 3) if isinstance(reg["value"], float) else reg["value"]`

Does this interfere with other values being rounded?

> [image]

Do you create a PR? 😊

ssunny1081, 2025-05-03:
Didn't know how to create a PR but seems quite easy. Please check if is OK like that.

Rouding seems not to affect any other Values. They are kept as is with less digits:

```
bat_sysstate: Discharging
tot_bat_soc_pct: 14
Vpv1: 43.14
Vpv2: 33.39
out_power: 436
out_voltage: 41.9
temp2: 29.31
maxcvbat1: 3.261
mincvbat1: 3.258
pv_tot_power: 145
```