robertzaage/GroBro #13: wifi credentials in mqtt stream and strored in dump?

vandecook, 2025-04-24:
Checking the registers of the NOAH2000 also was able to read in clear text the wifi credentials.

- Are this credentials also sent by the NOAH2000 with the MQTT stream to their servers?
- If yes, are these data also stored in the dump or is this content hidden in any way?

robertzaage, 2025-04-24:
> Checking the registers of the NOAH2000 also was able to read in clear text the wifi credentials.
> 
>     * Are this credentials also sent by the NOAH2000 with the MQTT stream to their servers?
> 
>     * If yes, are these data also stored in the dump or is this content hidden in any way?

It's not hidden or encrypted in any special way, but until now, I had never seen a device send its Wi-Fi credentials without the user explicitly reading the registers. If it's not read by user (or Growatt Cloud) it will not show up in a dump.

vandecook, 2025-04-24:
Thanks for your fast response.

closed 2025-04-24 by vandecook