robertzaage/GroBro #15: Default Werte Data Logger

nschaefer, 2025-04-24:
Hi,

ich würde gerne ausprobieren, ob ich GroBro zum Laufen bekomme.
Ich habe soweit alles aufgesetzt.

Das Einzige was noch fehlt, sind jetzt die Datalogger Settings des Noah auf den lokalen Mosquitto zu setzen.
Ich zögere jetzt nur noch, weil ich die Default Werte für die Domain und den Port nicht kenne und ich sie auch im Webinterface nicht einsehen kann.

Wenn ich jetzt den lokalen mosquitto setze und das Ganze nicht funktioniert, wie komme ich dann zurück auf die Default Werte?
Ist es "server.growatt.com"? Und welcher ist der Default Port?

Vielen Dank schon mal und viele Grüße

vandecook, 2025-04-24:
Hi,
ein we ist, du setzt die Einstellungen vom NOAH zurück (iot Taste mindestens 5 Sekunden drücken glaube ich) und bindest ihn über die App neu ein.

nschaefer, 2025-04-24:
Vielen Dank für eure schnellen Antworten.

Wenn ich mir das jetzt so anschaue, würde ich bei Problemen gar nicht mehr über das Webinterface den Data Logger auf Default Werte zurücksetzen können, da er dann nicht mehr erreichbar wäre.
Also bleibt dann eigentlich nur der Reset über die IOT Taste und neu hinzufügen, richtig?