robertzaage/GroBro #16: MQTT Connection, aber keine Daten

nschaefer, 2025-04-25:
Hi,

leider funktioniert das Ganze bei mir noch nicht richtig.

Ich habe jetzt folgenden Aufbau:

SOURCE MQTT:
Auf Synology NAS in Docker Container

GroBro:
Addon auf HomeAssistant.

TARGET MQTT:
HomeAssistant Mosquito

DNS Server:
PiHole auf Synology NAS (Docker Container)

Ich sehe jetzt in den Logs der verschiedenen MQTT Server, dass Verbindungen zustande kommen, aber ich erhalte keine Daten in HomeAssistant.
Ich hänge mal nachfolgend Auszüge der Logs an:

SOURCE MQTT (von unten nach oben):
2025/04/25 14:52:54	stderr	1745585574: Sending PUBACK to <SN_Nummer> (m6009, rc0)
2025/04/25 14:52:54	stderr	1745585574: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:54	stderr	1745585574: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6009, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:49	stderr	1745585569: Sending PUBACK to <SN_Nummer> (m6008, rc0)
2025/04/25 14:52:49	stderr	1745585569: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:49	stderr	1745585569: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6008, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:44	stderr	1745585564: Sending PUBACK to <SN_Nummer> (m6007, rc0)
2025/04/25 14:52:44	stderr	1745585564: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:44	stderr	1745585564: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6007, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:39	stderr	1745585559: Sending PUBACK to <SN_Nummer> (m6006, rc0)
2025/04/25 14:52:39	stderr	1745585559: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:39	stderr	1745585559: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6006, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:34	stderr	1745585554: Sending PUBACK to <SN_Nummer> (m6005, rc0)
2025/04/25 14:52:34	stderr	1745585554: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:34	stderr	1745585554: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6005, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:29	stderr	1745585549: Sending PUBACK to <SN_Nummer> (m6004, rc0)
2025/04/25 14:52:29	stderr	1745585549: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:29	stderr	1745585549: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6004, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:25	stderr	1745585545: Received PUBACK from <SN_Nummer> (Mid: 1, RC:0)
2025/04/25 14:52:25	stderr	1745585545: Sending PUBACK to <SN_Nummer> (m6003, rc0)
2025/04/25 14:52:25	stderr	1745585545: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:25	stderr	1745585545: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6003, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:25	stderr	1745585545: Sending PUBACK to grobro-source (m14, rc0)
2025/04/25 14:52:25	stderr	1745585545: Sending PUBLISH to <SN_Nummer> (d0, q1, r0, m1, 's/33/<SN_Nummer>', ... (67 bytes))
2025/04/25 14:52:25	stderr	1745585545: Received PUBLISH from grobro-source (d0, q1, r0, m14, 's/33/<SN_Nummer>', ... (67 bytes))
2025/04/25 14:52:25	stderr	1745585545: Sending PUBACK to <SN_Nummer> (m6002, rc0)
2025/04/25 14:52:25	stderr	1745585545: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (411 bytes))
2025/04/25 14:52:25	stderr	1745585545: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6002, 'c/33/<SN_Nummer>', ... (411 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:22	stderr	1745585542: Denied PUBLISH from grobro-source (d0, q0, r0, m0, '$SYS/33/state', ... (1 bytes))
2025/04/25 14:52:21	stderr	1745585541: Sending PINGRESP to <SN_Nummer>
2025/04/25 14:52:21	stderr	1745585541: Received PINGREQ from <SN_Nummer>
2025/04/25 14:52:21	stderr	1745585541: Sending PUBACK to <SN_Nummer> (m6001, rc0)
2025/04/25 14:52:21	stderr	1745585541: Sending PUBLISH to grobro-source (d0, q0, r0, m0, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:21	stderr	1745585541: Received PUBLISH from <SN_Nummer> (d0, q1, r0, m6001, 'c/33/<SN_Nummer>', ... (331 bytes))
2025/04/25 14:52:21	stderr	1745585541: Sending PINGRESP to <SN_Nummer>
2025/04/25 14:52:21	stderr	1745585541: Received PINGREQ from <SN_Nummer>
2025/04/25 14:52:21	stderr	1745585541: Sending SUBACK to <SN_Nummer>
[... 45 lines cut ...]
2025-04-25 14:52:34,308 [INFO] Published state for <SN_Nummer> with 0 registers
2025-04-25 14:52:39,388 [INFO] Device <SN_Nummer> matched 0 registers after filtering.
2025-04-25 14:52:39,388 [INFO] Published state for <SN_Nummer> with 0 registers
2025-04-25 14:52:44,312 [INFO] Device <SN_Nummer> matched 0 registers after filtering.
2025-04-25 14:52:44,313 [INFO] Published state for <SN_Nummer> with 0 registers
2025-04-25 14:52:49,327 [INFO] Device <SN_Nummer> matched 0 registers after filtering.
2025-04-25 14:52:49,328 [INFO] Published state for <SN_Nummer> with 0 registers
2025-04-25 14:52:54,338 [INFO] Device <SN_Nummer> matched 0 registers after filtering.
2025-04-25 14:52:54,338 [INFO] Published state for <SN_Nummer> with 0 registers
2025-04-25 14:52:59,354 [INFO] Device <SN_Nummer> matched 0 registers after filtering.
2025-04-25 14:52:59,354 [INFO] Published state for <SN_Nummer> with 0 registers
2025-04-25 14:53:04,314 [INFO] Device <SN_Nummer> matched 0 registers after filtering.
2025-04-25 14:53:04,314 [INFO] Published state for <SN_Nummer> with 0 registers
2025-04-25 14:53:09,294 [INFO] Device <SN_Nummer> matched 0 registers after filtering.
2025-04-25 14:53:09,294 [INFO] Published state for <SN_Nummer> with 0 registers

TARGET MQTT:
2025-04-25 14:51:19: New connection from 172.30.32.1:56930 on port 1883.
2025-04-25 14:51:19: New client connected from 172.30.32.1:56930 as grobro-target (p2, c1, k60, u'<User>').

Wenn ich das richtig deute, kommen zwar MQTT Nachrichten am GroBro an, aber keine Daten. Dementsprechend habe ich auch keine Entitäten in HomeAssistant.
Ist das ein Problem mit den Zertifikaten? Würde sich ein Problem mit den Zertifikaten so darstellen?
Mit der Erstellung der Zertifikate hatte ich ohnehin meine Schwierigkeiten und ich habe keine Ahnung, ob die passen.

Vielen Dank schon mal und viele Grüße.

robertzaage, 2025-04-25:
Ggf. ist dein Problem ganz simpel.

> 2025-04-25 14:53:09,294 [INFO] Device <SN_Nummer> matched 0 registers after filtering.
2025-04-25 14:53:09,294 [INFO] Published state for <SN_Nummer> with 0 registers

Kannst du mal die `REGISTER_FILTER` Option auf die Seriennummer deines NOAHs setzen? Etwa so: `123456789:NOAH`

Dann weiß die Bridge welches Mapping sie verwenden muss.

nschaefer, 2025-04-25:
Ach sehr cool. Das hat funktioniert. 
Vielen lieben Dank!

closed 2025-04-25 by robertzaage