robertzaage/GroBro #22: False values for Neo-800 Inverter

jojeji, 2025-04-28:
@robertzaage  since 1.6 I have invalid values for my neo-800 inverter:

[image]

Please let me know if you need more details and I provide it :)

robertzaage, 2025-04-28:
What's your current GroBro config and are these strange values stay forever or do they show up occasionally?

robertzaage, 2025-04-28:
And what's your firmware version? Growatt pushed an update for NEOs two days ago.

jojeji, 2025-04-29:
strange... it seems to work now. I just restartet GroBro and reconnected the Inverter and now the values are correct. But these steps I did several times the last days with no success.

[image]
#Closed

closed 2025-04-29 by jojeji