robertzaage/GroBro #24: Neo verliert manuelle mqtt Einstellungen über Nacht / Neo loses manual mqtt settings overnight

ma-schmitz, 2025-04-29:
Hallo,
mein Neo verliert die mqtt Einstellungen über Nacht - woran kann es liegen?
Mal funktioniert es, mal leider nicht und er verliert die Einstellungen.

Hello,
my Neo loses the mqtt settings overnight - what could be the reason?
Sometimes it works, sometimes it doesn't and it loses the settings.

1745866064: Client QMN000xxxxxxxxxx closed its connection.
1745866064: OpenSSL Error[0]: error:0A000126:SSL routines::unexpected eof while reading
1745862831: New client connected from 172.17.0.1:47146 as grobro-source (p2, c1, k60).
1745862831: New connection from 172.17.0.1:47146 on port 7006.
1745862468: Client grobro-source closed its connection.
1745862468: OpenSSL Error[0]: error:0A000126:SSL routines::unexpected eof while reading
1745850324: Client mqtt-explorer-593f8537 disconnected.
1745837123: New client connected from 172.17.0.1:50222 as grobro-source (p2, c1, k60).
1745837123: New connection from 172.17.0.1:50222 on port 7006.
1745837118: Client grobro-source closed its connection.
1745837118: OpenSSL Error[0]: error:0A000126:SSL routines::unexpected eof while reading
1745836954: New client connected from 172.17.0.1:49850 as QMN000xxxxxxxxxx (p2, c0, k420, u'QMN000xxxxxxxxxx').

robertzaage, 2025-04-30:
What's your current GroBro configuration? Do you use the cloud forwarding feature and/or the shinephone app? Is your MQTT setup correct - mind DNS problems with dynamic IPs if you don't rewrite the record it in your home network.

ma-schmitz, 2025-05-02:
Hi,
I use cloud forwarding and I use also the shine phone App.
I will try the DNS rewrite.

ma-schmitz, 2025-05-05:
Hello,
unfortunately my NEO still loses its settings overnight.
I have deactivated cloud forwarding - unfortunately no change.
So the problem should not be with the Growatt servers.
I had a similar behavior once with a ShineWiFi stick.
If it could not reach the server, it always loaded the default settings.
I don't think it's a real GroBro problem - but do you have any ideas?
What's strange is that I have to change the server and the port online on the website - it doesn't accept the values with the app.

robertzaage, 2025-05-05:
> Hello, unfortunately my NEO still loses its settings overnight. I have deactivated cloud forwarding - unfortunately no change. So the problem should not be with the Growatt servers. I had a similar [...]

Maybe you use a wifi mesh system? The ESP-C3 in the NEO has problems with those.

> What's strange is that I have to change the server and the port online on the website - it doesn't accept the values with the app.

You need to press save after changing the values in the app and then go one back and apply the settings. This restarts the datalogger too. (See configuration part in this repo)

ma-schmitz, 2025-05-05:
> Maybe you use a wifi mesh system? The ESP-C3 in the NEO has problems with those.

My WiFi Setup:
FritzBox with disabled 2,4 Wifi and enabled 5 Wifi.
Unify Access Points 2,4 WiFi only.

> You need to press save after changing the values in the app and then go one back and apply the settings. This restarts the datalogger too. (See configuration part in this repo)

ok, i will try it.

gfelbing, 2025-05-18:
Can reproduce, my NEO lost its connection too last night.
In my case, the inverter was never connected to growatt cloud: I blocked it's WAN access in my router and directly configured my non-forwarding grobro instance with the first startup.

My current theory, that this happens in the following case:

- DC (solar) input drops to zero
- mqtt is down when DC input comes back up

I'll make sure the mqtt instance is up tomorrow morning, let's see.

gfelbing, 2025-05-24:
After further testing: My theory holds, when NEO boots and it cannot connect to the mqtt, it seems to reset its options.
So the proper workaround would be to make sure that you're mqtt server is available in the morning :)
I guess adding a NOAH would also fix the issue, then NEO probably won't reboot at all.
Otherwise I doubt that we can do anything about it, as grobro cannot set the mqtt port without mqtt connection