robertzaage/GroBro #29: Help offer

gfelbing, 2025-05-02:
Hi there!
Thank you for your awesome work so far on this project!
I am currently waiting for the delivery of my noah 2000 / neo 800 m-x and I am looking forward to use this to cut it from the cloud and integrate it into my HA.

Looking forward to control the system, especially the output power of the system.
I see that you are already actively working on that, is there anything where I can support?

robertzaage, 2025-05-02:
Thank you for the offer, I really appreciate this! I'm currently a little short of time and there is some heavy lifting to be done on creating something useful (like a library) from all we got so far. 😊

See my comment here: https://github.com/robertzaage/GroBro/pull/28#issuecomment-2845279549

TLDR; Refactoring needed and a HACS integration would be my goal.

gfelbing, 2025-05-02:
Yes, the HA integration would definitely be a thing. 
It's a bit hard to test w/o having the hardware at hand (yet), but I'll try to mock sth up later.
If that works out, you'll see PRs :)

robertzaage, 2025-05-02:
Does it make sense to create something like a discord channel for all who want to contribute to the project?

gfelbing, 2025-05-02:
Great idea, I'd be in favor of sth more open though, e.g. matrix?
Could be sth like this: https://matrix.to/#/!CQCbGtMNlggCnXgwdl:matrix.org?via=matrix.org

robertzaage, 2025-05-02:
Alright, I joint the room 😊 I would add the link to the readme later.

criticallimit, 2025-05-02:
I wonder if the Noah-MQTT can be used with GroBro....

gfelbing, 2025-05-02:
Do you mean https://github.com/mtrossbach/noah-mqtt?
I don't think that would be a problem, as noah-mqtt sits behind the growatt cloud. If you enable ACTIVATE_COMMUNICATION_GROWATT_SERVER then all the data is still there

criticallimit, 2025-05-02:
can you please give a hint how you managed to avoid the cloud using noah-mqtt?