robertzaage/GroBro #33: HA-Guide Question

criticallimit, 2025-05-03:
In the guide I miss the example for what to insert here (see picture).

Is it  xxxxxxxxx.duckdns.org ?
Or, if I use it local only, can I put the IP of HA here (192.168.178.10) ?

[image]

joni347, 2025-05-03:
This must be the address for which you created your TLS certificate.

criticallimit, 2025-05-03:
ok. But I created the certicate with duckdns and before I added the local IP instead off the global IP.
Once I created the certifcates I don´t need duckdns any longer, because it has the local IP before the certificates are built.

The question is, can I use the local IP now instead of xxxxxxx.duckdns.org to enter in that field?

joni347, 2025-05-03:
You'll have to try it, but I don't think it will work. Normally the IP is checked against the certificate when the connection is made. The certificate also expires at some point and needs to be renewed.

joni347, 2025-05-03:
I think the best way to stay completely local is to set the Duckdns IP in Noah and redirect the Duckdns IP to your local IP in your router. As described in the instructions.

coor, 2025-05-23:
Hi,
I was not able to change the mqtt setting in the Shine App. Finally I changed the server address using the cloud webserver modifying register 17. All seem to work fine. Luckyly noah2mqtt still works. The Growatt web interface still updates values.

Leider, I did not find the Advanced Button in the Shine App to change the mqtt server. Can you please explain how to get there, because it is now not possible to change the mqtt-server/registers using the web interface any more.   

[image]

[image]

vandecook, 2025-05-23:
Is GROWATT_CLOUD enabled? If yes, it should be possible to modify with the web interface.

joni347, 2025-05-23:
> Leider, I did not find the Advanced Button in the Shine App to change the mqtt server. By the way I use a HM1500 inverter. Can you please explain how to get there, because it is now not possible to change the mqtt-server/registers using the web interface any more.

You must switch off the WiFi on the Noah (IoT button). Then go to GroHome in the ShinePhone app. If the Noah is displayed as offline and you click on it, you can reconfigure the network. Under the settings for the WLAN is then the button “Advanced”. There you can set the MQTT server. The password for the advanced settings is “growattYYYYYMMDD” (YYYYMMDD -> today's date)