robertzaage/GroBro #34: Inverter goes offline without MQTT update, HA shows stale values

ncwoehler, 2025-05-05:
**What's the issue?**

When my Noah 2000 battery runs out of energy, the Neo-800 inverter powers down but doesn't send a final MQTT update before going offline.
As a result, Home Assistant continues to display the last reported value, which can be misleading until the system wakes up again.

**Example:**
At night, the Noah 2000 discharges completely, and both output and discharge power drop to 0W.
However, HA still displays the last reported 129W from the Neo-800, even though it's offline, until sunrise reactivates both the battery and inverter.

[image]

**Expected behavior**
Ideally, HA should know that the inverter has gone offline or stopped reporting, so it can mark the data as stale or unavailable.

**Suggestion**
I know GroBro just forwards the data it receives, but would it make sense to add basic state tracking—like a timeout or heartbeat check—to flag when a device stops reporting?

robertzaage, 2025-05-06:
Thank you for reporting and @jojeji thanks for the quick fix. I just merged the PR. 😊

ncwoehler, 2025-05-06:
Wow, @jojeji - that was fast! I was planning to dig into it today and possibly open a PR myself, but looks like you’ve already nailed it 😄👏 Thanks for the quick fix!

closed 2025-05-11 by robertzaage