robertzaage/GroBro #38: Preventing NOAH from switching back to mqtt.growatt.com

msmthng, 2025-05-06:
what is the best way to preventing noah from switching back to growatts mqtt server?

does anybody know if noah has a register with a fallback value for the mqtt server. after the noah was offline yesterday it switched back to the official server. for sure it could also be that there was an update of register 17 sent my growatt - can we block this request in bridge mode?

robertzaage, 2025-05-06:
Yes, blocking (specific) register writes is a feature I think is really important. My NEO is stuck in OTA update mode because I blocked it from internet but allow it to communicate with the Growatt cloud. They send a OTA link and now it has a hard time trying to download the package 😄 No real problem, but at least annoying.

msmthng, 2025-05-06:
and maybe we should be able to overwrite values send to growatt e.g. if they are reading out register 17 we should respond with the value they have set?

ssunny1081, 2025-05-18:
Today after about 1 month, 1 out of 2 Noah was not connected to the WLAN, shady weather and was already empty yesterday evening. Connection was established again by just pressing the IOT button.

The ugly but, MQTT setting was lost.
Wrote Register 17 to the local MQTT -> not connect to GroBro
Restart Datalogger -> not connected
Again up on the roof, IOT off / On -> not connected
Noah off / on -> not connected
connected via shineapp, set local MQTT and Wifi -> not connected
Noah on / off not connected
Reset 5sec IOT Button -> setup again with local MQTT -> finally got connected again

As a Feature Idea:
Would it be possible that the componend forwarding the Data to Growatt could use its own DNS Server?

I would point the DNS of the official growatt mqtt address to a local reverse Proxy and than redirect to the local MQTT address, that the certificate is valid?
So the NOAH would always connect local, but the fowarding componend would still have the need to get the offical IP.

BTW: 
While I was looking for a way to automatically renew the certificate chain, I realized that the connection did work without the root and also without root & intermidediate.

During the reconnecting procedure described above I first thought its not working, but also with the full chain reconnect did not work without reset.

So in the end after connection reset, the NOAH connected again to GroBro without the Full Chain.

I use HAProxy on pfSense as TLS termination point, so refresh is simply done by ACME (O=Let's Encrypt, CN=E5, C=US) every 3 month.

msmthng, 2025-06-13:
is there a update on preventing unwanted updates to specific registers (not forwarding all requests from growatt?)

robertzaage, 2025-12-15:
Now available in `dev` for testing and will come next release.

closed 2025-12-15 by robertzaage