robertzaage/GroBro #43: Shelly integration

JanSchmidt3, 2025-05-08:
does anybody know, how the noah 2000 gets the power from the shelly?

I have one shelly and two noahs (parallel). Regarding noah documentation this setup won't work. But: I disabled the shelly cloud and enabled MQTT. The noah fount the shelly with autodiscover and could connect, which means there is a direct local connection without any cloud between the shelly and the noah.

The Idea is to create something like a wrapper script to which both noahs can connect to. This wrapper reads the current power from the shelly and sends it to the noahs - to each the half.

ssunny1081, 2025-05-08:
Also the setup looks differently lately. Shineapp does not ask for the shelly cloud account anymore.

If it is realy local meanwhile, maybe Unimeter could be used [https://github.com/sdeigm/uni-meter](url)

Issue might be if one noah could deliver more power than the other. Than just Splitting is not sufficient.

joni347, 2025-05-09:
In previous versions, Noah got the data from Shelly via Shelly Cloud -> Growatt Cloud -> Noah.
I think in the current version it searches the local network for the webserver of the Shelly and gets the data directly from the webpage, just like the HA integration for the Shellys.
I wanted to test it yesterday, but unfortunately my Noah can't find the Shelly on the local network and integration via the Shelly cloud is no longer offered.

closed 2025-05-17 by robertzaage