robertzaage/GroBro #44: Neo Inverter: Today generated energy not showing up in Home Assistant

frankd2112, 2025-05-08:
in growatt_inverter_registers.json today generate energy is defined but it doesn't show up as entity in Home Assistant. Is this a limitation from the inverter itself? I'm with the latest dev version.

`{
      "register_no": 3049,
      "variable_name": "Eac today",
      "description": "Today generate energy",
      "unit": "kWh",
      ...
`

If you need logs or anything else please let me know.

frankd2112, 2025-05-08:
`2025-05-08 19:32:27.482 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Load Percent/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.145 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Inverter Status (mode)/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.150 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Inverter Status/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.287 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Ptouser total/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.288 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Ptogrid total/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.289 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Ptoload total/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.289 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Time total/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.290 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Eac today/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.291 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Eac total/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.364 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Derating Mode/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.365 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_PV ISO value/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.423 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_P Bus Voltage/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.424 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_N Bus Voltage/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.442 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Fault code/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.444 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Warning code/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.447 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Systemfault word0/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
[... cut ...] wed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
2025-05-08 19:33:27.542 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Received message on illegal discovery topic 'homeassistant/sensor/grobro/QMN000BZP3N7G171_Load Percent/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
`

seems like home assistant doesn't like spaces in the topics

robertzaage, 2025-05-11:
Looks like a regression after refactoring the code base. Can you check if https://github.com/robertzaage/GroBro/commit/e2f14876863f01b0e28dbe23d0f7927fba11ebd4 fixed this on `dev`?

closed 2025-05-11 by robertzaage

frankd2112, 2025-05-11:
@robertzaage, thank you it is working now. 

One strange thing I observe now: The daily energy production is giving me wrong results. I seems to me that it doesn't reset at the end of day and is accumulating over the days since I cut the internet connection for the inverter. Could be that there are some messages from the cloud to the inverter that trigger the daily reset.

PR #49 (2025-05-11): Remove whitespaces from variable_names and add validation
Fixes #44