robertzaage/GroBro #46: creation of new version

vandecook, 2025-05-09:
Normally on this project I work with the dev version because a lot of new functions are added which can be used.

Today I tried to update the image in portainer, and now i can not use the dev version any more. I also tried to create a new container from the dev. Starting ended with "exited - code 255"

Log: exec ./run.sh: no such file or directory

May it be possible to create a new version (1.7) with the latest features (e.g. Charging Power (NOAH2000)) 

Thanks 
Roland

frankd2112, 2025-05-09:
I had the same problem with the run.sh not being found. Have a look here: https://github.com/robertzaage/GroBro/discussions/36

vandecook, 2025-05-09:
do you also use portainer? 
I'm not very familiar with Docker and Portainer yet,, so currently it´s more trail and error.

vandecook, 2025-05-09:
Thank you. 
I now solved it in portainer.

[image]

closed 2025-05-09 by vandecook