robertzaage/GroBro #50: Current  Version does not start anymore v1.7.1

ssunny1081, 2025-05-11:
Just updated to the :latest build, now the container does not start anymore `exec ./run.sh: no such file or directory`

Thank you for checking

SohnyBohny, 2025-05-11:
Seems related to #46. But even if I set the exec command manually to `python -m grobro.ha_bridge` I get the following error:

```
Traceback (most recent call last):
  File "<frozen runpy>", line 189, in _run_module_as_main
  File "<frozen runpy>", line 112, in _get_module_details
  File "/app/grobro/grobro/__init__.py", line 1, in <module>
    from .client import Client
  File "/app/grobro/grobro/client.py", line 13, in <module>
    import paho.mqtt.client as mqtt
ModuleNotFoundError: No module named 'paho'
Traceback (most recent call last):
  File "<frozen runpy>", line 189, in _run_module_as_main
  File "<frozen runpy>", line 112, in _get_module_details
  File "/app/grobro/grobro/__init__.py", line 1, in <module>
    from .client import Client
  File "/app/grobro/grobro/client.py", line 13, in <module>
    import paho.mqtt.client as mqtt
ModuleNotFoundError: No module named 'paho'
Traceback (most recent call last):
  File "<frozen runpy>", line 189, in _run_module_as_main
  File "<frozen runpy>", line 112, in _get_module_details
  File "/app/grobro/grobro/__init__.py", line 1, in <module>
    from .client import Client
  File "/app/grobro/grobro/client.py", line 13, in <module>
    import paho.mqtt.client as mqtt
ModuleNotFoundError: No module named 'paho'
```

SohnyBohny, 2025-05-11:
Setting the exec command to `/venv/bin/python -m grobro.ha_bridge` worked for me.

ssunny1081, 2025-05-11:
Using docker / portainer, working for me as well, Thanks for the hint.

gfelbing, 2025-05-11:
@SohnyBohny While the fix works as long as you configure your container directly, it ignores configuration values from HA when you use it as addon.
The root cause is: The shebang line of the run.sh points to bash, but the alpine image doesn't have bash installed. I fixed it in #51

PR #51 (2025-05-11): Fix docker build
- fix shebang line of run.sh
- add missing dependency

fixes #50
closed 2025-05-11 by robertzaage