robertzaage/GroBro #55: Confusion: target TLS defaults to source TLS when specified

LGro, 2025-05-15:
I just stumbled across an issue where I set `SOURCE_MQTT_TLS=true` without setting `TARGET_MQTT_TLS`, assuming that means `TARGET_MQTT_TLS=false`. However, this turned out to not be the case, because the [target config defaults to the source config](https://github.com/robertzaage/GroBro/blob/main/grobro/ha_bridge.py#L45).

This seems to be intended, so I'll leave it to you to judge whether this warrants a change or not; Just wanted to let you know it didn't match my expectations.

robertzaage, 2025-05-17:
@gfelbing @jojeji Do we need to be more clear on this in our docs or shall we handle this differently?

jojeji, 2025-05-17:
@robertzaage mhm I think we should be more clear in documentation.

LGro, 2025-05-17:
To be fair, once I read the documentation it was clear. My confusion originated from the mindset I had before reading - that these would be independent things and that in the locally self hosted space no TLS would be the default. Which is interesting to me, because generally I'd assume with TLS to be the default.

So if I see potential for improvement: make them both mandatory to avoid needing to deal with assumptions about defaults and instead force explicitness. The docs are fine, I'd say.

gfelbing, 2025-05-18:
TBH, I just fell for it too :D 
I'm with @jojeji , I'd keep the behavior, but we should be more transparent.
IMHO we should

- make it more obvious in the docs
- print the configuration at startup (except secrets, of course)

Latter one would need a bit of refactoring for the config in general, e.g. moving to ArgumentParser

robertzaage, 2025-08-09:
We should check certificate validity too and warn users if their cert is expired or self signed. 😊