robertzaage/GroBro #56: Only receiving nonsene over MQTT

REDDE4D, 2025-05-16:
Hello,
after a long chain of problems i finally managed that my NEO 800M-X connected to my TLS MQTT Brocker.
My problem now is that i only receive nonsense data like:

Topic:
`c/33/Inverter-Logger SN`

Message:
```

A?!GQD6"Y'P5D$owattGrowattGr--1B$v3_$7ttGrowattGrowattGrowxqdJ^huj�xsrnwajQFKo
at{GsTwttHWowattGrowattGrowattGrowattZf|�h<tXroj~ttGrowattGrowattN:owattGrowattGrowawYrow}ttGowatGrawatBGroyattqrowattGrowattGrowattGrowattGrowattGrowattGrswattN�owattGrowau�F�n�`�tGcwa:lG/owattGrowattGrowattGrowattG3owattGro'GttGrowatxr~�wattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowattGrowhg�Gr@�autG1�
```

Anyone know what the problem could be?

Thanks ahead
REDDE4D

LGro, 2025-05-17:
How did you configure the `REGISTER_FILTER` (see [here](https://github.com/robertzaage/GroBro/blob/main/CONFIGURATION.md#environment-variable-reference))?

robertzaage, 2025-05-17:
On the latest `dev` release setting REGISTER_FILTER is not needed anymore, as we're auto configuring the parser by device serial.

As @LGro already explained GroBro reads this "nonsense" and republishes the data in a form Home Assistant understands. You're on the right track! 😊

REDDE4D, 2025-05-18:
It worked out, fixed itself. Had it running over night and in the morning there where finally data in Homeassistant.
Thanks :)

closed 2025-05-18 by REDDE4D