robertzaage/GroBro #93: No data every new month

SohnyBohny, 2025-08-02:
Thank you for GoBro! I'm a happy user since the beginning.

Every new month (i.e. August 1st, Junly 1st, June 1st) GoBro stops sending data to Home Assistant. I already noticed this the last two months but I kinda thought it was random and fixed it by restarting the GoBro Docker container.

Did anyone else observe something similar? How can I debug this or provide logs?

(I was always running the latest stable version of GoBro, so currently I'm running v2.0. My setup is local and my inverter is blocked from the internet so doesn't know the current date.)

SohnyBohny, 2025-08-02:
MQTT Explorer indicates that GroBro is still a "connected" and "active" MQTT client on both source and target brokers and my inverter is still sending messages on the source broker (while grobro is not forwarding them to the target broker).

I mean, this bug is not a big issue since I was able to solve it by restarting the docker container in the past. I haven't done so this month yet because I wonder if someone has experienced similar behavior or has an idea on how I could debug the current zombie state?

SohnyBohny, 2025-08-02:
When I press the new "read all values" button in home assistant a message on the `s/33` topic is sent to the inverter (which in turn responds with a `c/33` message). But still no messages / updates are sent to home assistant. So GroBro seems to still work in one direction but not in the other.

robertzaage, 2025-08-02:
This bug is really strange… was the mosquitto container updated and therefore restarted? Looks like a connection stalled or something. 

Could you try restarting your MQTT broker and check if this triggers the issue?

SohnyBohny, 2025-08-02:
You were right, thank you! 🎉
My mosquitto is restarting once a month. GroBro also successfully reconnects after a restart but does not subscribe to the topic again. I proposed #94 to fix this.

PR #94 (2025-08-03): Subscribe on_(re)connect
This fixes #93 that GroBro does not resubscribe if the connection to the broker is interrupted and reestablished.
closed 2025-08-03 by robertzaage