robertzaage/GroBro #118: No way to change MQTT of NOAH battery on iOS

crazyracer98, 2025-09-04:
Hello everyone,

I wanted to ask if someone managed to change the MQTT server in the ShinePhone app on iOS?
If not, this should be included in the docs that this is not possible and some other workaround should be found and documented.

Many thanks in advance.

crazyracer98, 2025-09-04:
Ok I will try that on the weekend! If that is the way to go, we should do a PR to the configuration docs page to include that for iOS users.

crazyracer98, 2025-11-15:
Hi, sorry, life got in between me and trying this out.
If I delete the NOAH device from within the ShinePhone app and try to re-add the device, I get the advanced setup menu options. Hurray! So far so good.
When I then go ahead and do the following config changes:
- WiFi Settings: DHCP -> enabled
- Server Settings: my-mqtt.my-local-domain / 7006

I do not see any connections in my MQTT server. I also tried it with Port 8883.
I have a local DNS server on my OPNsense running, which resolves `my-mqtt.my-local-domain` correctly.
My Mosquitto has the following setup:
```conf
per_listener_settings true

listener 1883 0.0.0.0
allow_anonymous true
protocol mqtt

listener 8883 0.0.0.0
allow_anonymous true
cafile /certs/ca.pem
certfile /certs/everything.pem
keyfile /certs/key.pem

listener 7006 0.0.0.0
allow_anonymous false
password_file /mosquitto/config/pws.txt
cafile /certs/ca.pem
certfile /certs/everything.pem
keyfile /certs/key.pem
```

Do you know if maybe the Growatt devices ignore the DNS server given by DHCP?

robertzaage, 2025-11-15:
> Do you know if maybe the Growatt devices ignore the DNS server given by DHCP?

Yes, at least the NEO firmware indicates that they use a DNS from China Telecom. If you don’t redirect all DNS requests to your firewall, this would circumvent your custom config.