robertzaage/GroBro #122: NOAH 2000 - Local Shelly 3EM Pro Integration loosing connection - restart datalogger

Baymax2909, 2025-09-14:
Hey everyone!

I have my Shelly 3EM Pro integrated in my NOAH with the „new“ local integration. Before I installed GroBro I could restart the datalogger when the integration lost the connection. After that the integration works fine.

with the local GoBro setup I can not restart the datalogger any more. Could we integrate the option to restart the datalogger?
Or is there an other solution to stabilize the Shelly <-> Noah integration?

This problem fixes itself sometimes over night as datalogger restarts there „naturally“.

See Discussion #120

robertzaage, 2025-11-07:
We talk about a local-only option here? Restarting the logger via ShineApp when cloud-forwarding is enabled still should work.

Baymax2909, 2025-11-10:
Sorry for the late response.

Yes, we talk about a local only variant. I have not tried to restart it with cloud forwarding because I do not use the cloud forwarding function.